Ultimaker/Cura #17894: Unable to set initial extruder offset of extruders that are not the initial extruder number

ansonl, 2024-01-04:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ultimaker 2+ DXU and newer UM S7

### Reproduction steps

1. Have some machine start g-code to prime both extruders and retract the extruder that is not `initial_extruder_nr` by `switch_extruder_retraction_amount`. This way the second primed extruder that is used later in the print does not ooze filament while the initial extruder starts printing.
2. Slice a two object print with the two extruders enabled and assigned.
3. View generated G-code

### Actual results

Both extruders start printing at extruder offset 0 after they are switched to with `T0`/`T1` the very first time.
I did not find a way to signal to the second unused extruder that it had been previously retracted by `switch_extruder_retraction_amount`mm. 

When `T1` prints for the first time, filament does not come out for the first 16mm of filament extrusion (if using the Ultimaker preset for `switch_extruder_retraction_amount`).

```
T1
G92 E0
M105
M109 S225
M104 T0 S170
G1 F1500 E-3
;MESH:NONMESH
G0 F7200 X122.39 Y96.369 Z1.2
G1 F600 Z0.2
G0 F7200 X133.53 Y96.369
G0 X134.696 Y87.366
;TYPE:SKIRT
G1 F1500 E0.62363
G1 F900 X134.698 Y128.573 E1.99418
G1 X134.473 Y129.561 E2.02788
G1 X133.954 Y130.475 E2.06284
G1 X132.781 Y131.592 E2.11671
G1 X132.549 Y131.732 E2.12572
```

The second time the extruders are used, Cura correctly primes the nozzle according to the set `switch_extruder_retraction_amount` since Cura retracted them by that amount previously. 

```
T1
G92 E0
M105
M109 S200
M104 T0 S170
G1 F1200 E13
G1 F600 Z0.4
;MESH
G0 F12000 X115.774 Y107.55
M104 S210
;TYPE:WALL-OUTER
G1 F1500 E16.62363
G1 X115.748 Y107.266 E16.62837
G1 X115.693 Y106.949 E16.63372
G1 X115.624 Y106.648 E16.63886
```

### Expected results

I expected that Cura would remember the extruder retraction in the machine start gcode or have a setting checkbox to hint to the slicer that the extruders have been retracted by `switch_extruder_retraction_amount`mm in the start gcode. 

[... 114 lines cut ...]
M400 ;finish all moves
G0 Z20 F2400
G0 X212 F7200
G0 Y70 F7200
G0 X217 Y50 Z0.5 F7200
M109 T{initial_extruder_nr} S{material_print_temperature, initial_extruder_nr} ; Wait for initial nozzle to reach temp
G92 E0
G0 Y9.4 Z0.5 E3 F7200
G2 X212 Y3.4 Z0.4 I-6 J0 F7200
M104 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr} ; Start heating to first layer temp
G0 X207 Y3.4 Z0.4 F7200
G2 X205 Y5.4 Z0.3 I0 J2 F7200
G0 X205 Y5.4 Z0.3 F7200
G1 Y10 F18000 ; break line
G92 E0
G1 Y50 E2 F1000 ; extrude line
G92 E0
G1 E-0.5 F1500 ; retract slightly
G1 Y100 F18000 ; break line
G92 E0
M355 S1 P255; Turn on case light brighter
;end of startup sequence
```

This start gcode tries to start the initially unused T1 extruder at -3mm so that it oozes minimally but does not miss too much extrusion when T1 is used the first time and T1 resets extrusion offset at 0mm.

ansonl, 2024-01-04:
Passing in `initial_extruder_nr` works successfully since the initial extruder seems to be determined in the slicing step. `initial_extruder_nr` is also used in multiple other existing printer definition start gcodes as well. 

This is an issue where Cura should have an option of setting an initial extruder preset for first use of an extruder. Users would be able to signal to Cura whether their start gcode has retracted the longer `switch_extruder_retraction_amount` and Cura would start the initial extruder offset accordingly. 

I think that this "initial offset based on start gcode retraction" problem did not come up as an issue for dual extruder printers earlier because people just assumed that it was taking a little while for the extruder to catch up and start extruding -- not realizing that Cura is starting from 0 offset and the filament is still being moving past the `switch_extruder_retraction_amount` length.

5axes, 2024-01-04:
initial_extruder_nr  is define in the **StartSliceJob.py**  file .. You can add your own initial_first_unused_extruder_nr  ... But this modification have do be done for every new release.

`result["initial_extruder_nr"] = CuraApplication.getInstance().getExtruderManager().getInitialExtruderNr()`
 `result["initial_first_unused_extruder_nr"] = stack.getProperty("extruders_enabled_count", "value") - (CuraApplication.getInstance().getExtruderManager().getInitialExtruderNr()+1)`

ansonl, 2024-01-05:
Ok, I'll try Cura 5.6 to get the start gcode math working the unused extruder number like you mentioned in the other issue.

I'm currently trying to solve the extruder offset problem with a Cura post processing plugin and would like to set the default value for the settings field to `switch_extruder_retraction_amount_0`. I looked at your plugin repo code (https://github.com/5axes/Cura-Postprocessing-Scripts/blob/491e26cf89222a91fac8244f05364269db753d78/AlterZhops.py#L44) and you were able to call `resolveOrValue` to populate a value in the settings field. However I tried to do the same with `switch_extruder_retraction_amount` and `switch_extruder_retraction_amount_0` and `resolveOrValue` does not return a valid key which leads to string to float error. Do you have any ideas for getting the value of `switch_extruder_retraction_amount` to populate a settings field as the default value?

ansonl, 2024-01-05:
I made the post processing script [here ](plugins/PostProcessingPlugin/scripts/SetInitialSwitchedExtruderOffset.py )and will see if the additional offset helps with the second extruder's initial use over my next few prints.

5axes, 2024-01-05:
there are three function define in the CuraFormulaFunctions.py

- `resolveOrValue`
- `extruderValue`
- `extruderValues`

`resolveOrValue` is used to get a Global parameter like `resolveOrValue('layer_height')`

`extruderValues` is used to obtain all the values of all extruders and is often used with `min`, `max `or `any` :  `max(extruderValues('wall_line_width_0'))`

In your case you should try extruderValue :  `extruderValue(0, 'support_interface_height')`

fieldOfView, 2024-01-05:
> I don't think you can pass the extruder number as a variable. ie :
>
> `M104 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}`

That certainly used to be possible, because that is how I originally implemented it. If that is no longer possible, than that should at least be publicly documented, and otherwise it is a bug.

5axes, 2024-01-05:
Ok tested in 5.3 and 5.4 it's ok, so yes software regression due to the modification in 5.6

`M104 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}`

result in **5.3 & 5.4** : `M104 T0 S200`
result in **5.6** : `M104 T0 S(200, 0)`

ansonl, 2024-01-07:
I tested getting the extruder based setting in Cura 5.6 and got the same result ats @5axes with the parentheses. 

I made another bug report for it since it is different from this issue's problem for setting the initial extruder position offset on first use. 
My post processing script linked above fixes the initial extruder position offset by looking for the first use of `Tx` where `x` is the extruder number and replacing the next `G92 E0` with `G92 E-{switch_extruder_retraction_amount}` so that the extruder properly pushes the filament back into the melt zone and primes after being switched away in another print or the start gcode.