Ultimaker/Cura #17896: Support Z distance not perfect in v5.6

AnandRaj-R, 2024-01-04:
### Cura Version

5.6

### Operating System

Windows 10 Pro

### Printer

Anet A8

### Reproduction steps

1. Enable support with z distance of 0.12mm for normal layer heights of 0.2mm
******************************************
(0.12mm giving best surface finish for four start lead screw or 8mm per one rotation because the four start lead screw also move 0.8mm so (0.08 + 0.12 = 0.2mm) no calculation error or no rounding-off while calculating layer height)
******************************************

### Actual results

If i set the support z distance is 0.12, uneven and unwanted layer heights are generated below the support z distance.
[image]
[image]
[image]

### Expected results

If we set the support z distance 0.12mm for the normal layer heights 0.2mm
The below changes we are expecting 
1. No changes in object layers height
2. 0.08mm for last support layer height
3. 0.12mm layer gap between support and objects
4. Bridge layer height 0.2mm
Ex.,
Suppose bridge in 50th layer now
Layer 51 ----- 0.2mm object 
Layer 50 ----- 0.2mm bridge
Layer 49 ----- 0.2mm for object 
Layer 49 ----- 0.12 mm gap between support and bridge 
Layer 49 ----- 0.08mm for support 
Layer 48 ----- 0.2mm object & support 

Three activities involved in the 49th layer
1st print the object at 0.2mm layer height
2nd  Print support at 0.08mm layer heights
3rd move z at 0.12mm and then 
print bridge at 0.2mm and print continue 0.2mm layer height 

### Add your .zip and screenshots here ⬇️

[Support 0.12mm.zip](https://github.com/Ultimaker/Cura/files/13828006/Support.0.12mm.zip)

GregValiant, 2024-01-05:
Thanks for the report.
Cura does not understand G2 and G3 so I can't preview your gcode file. You haven't included a project file which is required for debugging.
I do see that there are a couple of layers at .08 layer height.  You have "Adaptive Layers" enabled and the .08 layer height may be a side effect of that.  This model does not require that Adaptive Layers be enabled.  Arc Welder is also not required.
I don't understand why you have support growing within your model in your screen shots.  I'm unable to reproduce that.
[image]

The normal situation is that the air gap between any support and the layer above it will be exact multiples of the layer height.  Cura will not adjust a part of a layer to some special layer height.
If you set the air gap above support interface to less than one layer height then the air gap will be "0".  If you set the air gap to over 1 layer height then it will be one layer high until you get to 2X layer height at which point it the gap will become 2 layer heights.  The same thing is true for Infill Layer Thickness and Support Infill Layer Thickness.  The only airgap that can be different than "layer height" is the airgap over a raft and the reason is that it is an entire layer, not just a portion of a layer.

If you turn off adaptive layers I believe you will see that .08 layer go away (but I can't duplicate it so I'm not sure).
I'll remove the bug label from this as it is expected behavior to not make adjustments to the layer height for portions of a layer.
You can put in a feature request if you like.  Post a project file and I can take another look.  I'll leave this open for a bit so you can respond.

AnandRaj-R, 2024-01-23:
I understood and thank you for your detailed explanation, thanks

GregValiant, 2024-01-23:
When layers are missing (or added where they shouldn't be) it is almost always a problem with the model and "inverted surfaces".  In this case that last STL you provided doesn't seem to have any errors and I can't duplicate the problem with my setup.
[image]
I would suggest you go back to a stock profile and start from there.  This is a simple model and should be fine.  If you continue to get missing layers then the overwhelming possibility is errors in the model.  MS 3D Builder has a pretty good repair capability and the app is available in the Microsoft Store.
If it was me, I'd do it this way.
[GV_Rod Mount Brkt LB2.zip](https://github.com/Ultimaker/Cura/files/14027241/GV_Rod.Mount.Brkt.LB2.zip)

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant