Ultimaker/Cura #17897: 5.6.0 will not slice something 5.2.0 has no problem slicing.

Zelious, 2024-01-04:
### Cura Version

5.6.0

### Operating System

Windows 10 Pro

### Printer

Ender 3 Pro

### Name abnormal settings

I'm using settings I've always used and they've never been a problem with previous versions. Its been so long since I changed them I don't know what isn't normal anymore. But yes, it does slice if I just choose one of the default settings. But I don't have to do that in Cure 5.2.0, my settings worked just fine then.

### Describe model location

It will not slice if I just move it from the center. But if I tilt it 90 degrees, it will slice. But that won't work for me.

### Describe your model

It is watertight according to the Meshtools plugin. I did have to fix it in 3d Builder, but that has never been a problem before. I just downloaded Cura 5.2.0 and it sliced it without a problem. This isn't the first time 5.6.0 hasn't been able to do something 5.2.0 did without a problem, so something changed for the worse.

### Add your .zip here ⬇️

[Drago_Basetta_B.zip](https://github.com/Ultimaker/Cura/files/13828059/Drago_Basetta_B.zip)

5axes, 2024-01-04:
An other "Print Thin Walls" issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452
[image]

Duplicate Slice Failed due to **Print Thin Walls**

https://github.com/Ultimaker/Cura/issues/17863
https://github.com/Ultimaker/Cura/issues/17750
https://github.com/Ultimaker/Cura/issues/17725
https://github.com/Ultimaker/Cura/issues/17679
https://github.com/Ultimaker/Cura/issues/17676
https://github.com/Ultimaker/Cura/issues/17671
https://github.com/Ultimaker/Cura/issues/17606
https://github.com/Ultimaker/Cura/issues/17560
https://github.com/Ultimaker/Cura/issues/17528
https://github.com/Ultimaker/Cura/issues/17503
https://github.com/Ultimaker/Cura/issues/17499
https://github.com/Ultimaker/Cura/issues/17470
https://github.com/Ultimaker/Cura/issues/17467
https://github.com/Ultimaker/Cura/issues/17457
https://github.com/Ultimaker/Cura/issues/17456
https://github.com/Ultimaker/Cura/issues/17496
https://github.com/Ultimaker/Cura/issues/17478
https://github.com/Ultimaker/Cura/issues/17439
https://github.com/Ultimaker/Cura/issues/17438
https://github.com/Ultimaker/Cura/issues/17410
https://github.com/Ultimaker/Cura/issues/17367
https://github.com/Ultimaker/Cura/issues/17344
https://github.com/Ultimaker/Cura/issues/17177
https://github.com/Ultimaker/Cura/issues/16688
https://github.com/Ultimaker/Cura/issues/16515
https://github.com/Ultimaker/Cura/issues/16417

MariMakes, 2024-01-18:
Hi @Zelious,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes