Ultimaker/Cura #17903: Slicing of a detailed high model failed

v0jk1, 2024-01-04:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

A10 Greeentech

### Describe model location

Center to high

### Describe your model

Model of an Chevrolet

### Add your .zip here ⬇️

[Uploading GA10_Chevrolet_impala.zip…]()

5axes, 2024-01-05:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

v0jk1, 2024-01-08:
The project data:
[GA10_Chevrolet_impala.zip](https://github.com/Ultimaker/Cura/files/13866597/GA10_Chevrolet_impala.zip)
And also the configuration:
[GA10_Chevrolet_impala.zip](https://github.com/Ultimaker/Cura/files/13866606/GA10_Chevrolet_impala.zip)

Thanks for your support. I thought i had upload it already

GregValiant, 2026-09-27:
I'm closing out old reports.
It's painted like a low rider.  That means a lot of model errors.  It just wasn't suitable for FDM as it seems to consist of surfaces rather than solids.

[image]

closed 2026-09-27 by GregValiant