Ultimaker/Cura #17905: Slicing failed with an unexpected error

renkam, 2024-01-05:
### Cura Version

5.6.0

### Operating System

Linux Debian 11

### Printer

Creality Ender-3 S1 Pro

### Name abnormal settings

It slices with vertical orientation, fails horizontaly

### Describe model location

center

### Describe your model

https://www.thingiverse.com/thing:2830825/files

### Add your .zip here ⬇️

[02PLAG_center_finder_tool_CE3S1PRO.3mf.zip](https://github.com/Ultimaker/Cura/files/13840354/02PLAG_center_finder_tool_CE3S1PRO.3mf.zip)

renkam, 2024-01-05:
Good to know, I've got only small error window, that tells me nothing.
[image]
It could print a message that support generation failed, so I could turn it off.

Thanks for tip!

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm here CuraEngine: Fix crashes caused by tree support.
CLOSED
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc