Ultimaker/Cura #17910: Tree Support Issue

cdyrud, 2024-01-05:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Wanhao Duplicator i3 Plus

### Name abnormal settings

Using tree supports, although I only changed the Z distance and changing it back did nothing

### Describe model location

Does not change if sliced in different corners of buildplate or at varying angles

### Describe your model

I have sliced it previously, it is watertight, I have not done a mesh fix

### Add your .zip here ⬇️

The issue seems to only be present if tree supports are needed. I turned off supports and the model sliced with no issue. I switched to normal supports and had no issues. I tried with other models and got similar results, although the issue is only consistent with the attached file for unknown reasons
[Bearing Fit.zip](https://github.com/Ultimaker/Cura/files/13845435/Bearing.Fit.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

cdyrud, 2024-04-06:
It slices correctly in 5.7. For whatever reason it's slicing just fine in
5.6 now too, but other models that don't are slicing correctly in 5.7 so it
seems to be solved. Thank you.