Ultimaker/Cura #17911: Trading Card box

Dickie102, 2024-01-05:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

eryone Thinker S

### Name abnormal settings

No

### Describe model location

No

### Describe your model

Not sliced before, not watertight, no option to install mesh fix on my account.

### Add your .zip here ⬇️

[ETS_Box_Simple.zip](https://github.com/Ultimaker/Cura/files/13846240/ETS_Box_Simple.zip)

GregValiant, 2024-01-05:
Thanks for the report.
It isn't watertight and it has other problems.  That being said, it sliced fine in 5.2.2, 4.13.1, and with Prusaslicer but not in later versions the latest versions of Cura.

The Cura team will take a look.

Dickie102, 2024-01-11:
Greg,

Many thanks for the feedback.

Regards
Des

GregValiant, 2024-01-11:
This sort of "No, I will not slice this" behavior has become more prevalent in Windows 11 systems.  Others are keeping tabs on these reports that seem to be Win11 specific.  I don't know where the Cura team is with this, but the number of reports is really growing as move people move to Win11.  I think it took everyone by surprise.

Dickie102, 2024-01-11:
Greg,

Many thanks and good to know.

Regards
Des

closed 2024-04-03 by HellAholic