Ultimaker/Cura #17915: Profiles and infill change at layer height

BelvederePixel, 2024-01-06:
### Is your feature request related to a problem?

Ive been using Cura for quite a bit ( not long in time, just extremely intensively lol ) and i came up to some frustration : When you try to chase a problem and you have to test so many different settings to get to a good final results, sometimes you dont think of writing down your changes. Sometimes you make it worse, sometimes you make it better. When you make it worse, but you changed the profile a couple times : it gets confusing to try to write down everything when youre running 9 printers, and troubleshooting different designs and prints at all times. 

### Describe the solution you'd like

So hear me out : what about a checkpoint system ? What about saving the changes of a profile and being able to access them. Maybe 5 of them ? maybe 10 ? 

 So lets say your profile is called "0.6mm nozzle" and you make changes (1) and you save the profile. Then you make changes (2) and save the profile. Then you test them, it works on one, but not the other. So you do more changes (3) and save them.

 

Now you realize the initial profile was the right one but youre too far gone. This is where my idea comes in :

 

You have your profile "0.6 nozzle" but beside it, under it, right click, whatever the option : you can open and select the previous states and have them listed as 1, 2, 3, 4, 5 etc etc.

 

Maybe im crazy, but with the amount of testing im doing for different things, different finish, different material, different shapes and colours, it would be very handy. 

### Describe alternatives you've considered

because i cant write everything for 9 STL on 9 printers. I get so confused ! lol

### Affected users and/or printers

Everyone that has short term memory problems and have been troubleshooting problems with their printers for 4 months straight, 8 hours a day lol

### Additional information & file uploads

the other thing is infill at layer height. I tried to use the gradient infill but it doesnt suit what i would like to do. 

I would want my bottom to be sturdy, but the infill in my walls doesnt need that much infill and my top needs supports for the roof : so if there was an option for infill % change at layer 6, my infill would be 30% from 1 to 6. Then 6 to 200 would be 10% and from 201 to 220 would be 20% lets say. Those are just random lines and numbers, but this is the idea.

GregValiant, 2024-01-08:
How about if the Cura settings for the slice were added to the file?  Curently, the last section of a Cura gcode contains the "Setting Changes".  Along with being hard to read, it only contains changes between the base stock profile (like "draft") and the settings of your Custom Profile.
I've been working on this post processor.  It will add the Cura settings to the end of the file.  It inserts all the Cura settings or you can select "by category".  Adding all the settings adds about 400 lines to the gcode for a single extruder printer.
It ends up looking like this in a text editor.
[image]

[AddCuraSettings_GV.zip](https://github.com/Ultimaker/Cura/files/13863539/AddCuraSettings_GV.zip)

Unzip the file and put "AddCuraSettings.py" into your Configuration Folder in the "scripts" sub-folder.  It will be available with the rest of the post-processors.
Let me know if it is helpful.

BelvederePixel, 2024-01-08:
well this is good ! the only issue is i just need to save the gcode everytime, since i send it to moonraker, and it doesnt save on my computer straight up. But i guess i can retrieve it from moonsail as well ! That does help ! thank you .... now what about the infill change at layer height or even any other settings ? lol is there a script for that that looks like a retracttower script but just for one specific settings and for the rest of the print ?

GregValiant, 2024-01-08:
For the infill changes you would need to use "support blockers" configured to "Modify Settings for Overlaps" and "Infill Mesh".  Once you have that you can select the changes you want for the overlapped area.  You might need a couple.  Usually Cura will allow denser infill at the top to hold up the roof.  For denser infill on the bottom you'll need a blocker.

This model has a blocker on the bottom that is larger than the model, configured with 0 walls, 0 tops, 0 bottoms, Infill Density 10%, Infill Line Multiplier at 3.  The rest of the model is per the regular settings and Infill Density at 5% and Infill Line Multiplier at 1.  The Infill Line Multiplier setting is much more efficient with 3 rather than 2.  It's a question of how the toolpaths are calculated.  Enabling "Connect Infill Lines" keeps retractions down to a minimum.
[image]