Ultimaker/Cura #17922: Tree Support Type Slice Fail

DarthSidneous, 2024-01-06:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Name abnormal settings

Support Type = Tree

### Describe model location

Rotate 90 Forward so flat edge it down = crash; Print vertically = no crash

### Describe your model

Small

### Add your .zip here ⬇️

[CE3PRO_DrawstringCap.zip](https://github.com/Ultimaker/Cura/files/13851377/CE3PRO_DrawstringCap.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

DarthSidneous, 2024-04-03:
Downloaded & Tested with the original model provided:
Test *Success *in each orientation.

Bravo! Thanks for looking into this, HellAholic

Best Regards,
DarthSidneous