Ultimaker/Cura #17924: Hole Horizontal Causes Crash

Moosiecakes, 2024-01-07:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3 V2

### Name abnormal settings

Added Value to Hole Horizontal Expansion. 0.583

### Describe model location

Center

### Describe your model

Globe Votive Winter Themed Ornament

### Add your .zip here ⬇️

[CE3E3V2_retromaker_ornament_globe3.zip](https://github.com/Ultimaker/Cura/files/13852292/CE3E3V2_retromaker_ornament_globe3.zip)

Moosiecakes, 2024-01-07:
When i removed the Hole Horizontal Expansion setting item would slice No problem. But as soon as i added it back it would not Slice and generate bug. 

Restarting Cura did not fix the issue.

GregValiant, 2024-01-08:
Thanks for the report.
There appears to be something going on in regards to Hole Horizontal Expansion.
When I set it to any non-zero number slicing appears to stop.  If I turn on "Print Thin Walls" then the slice finishes.  That's your workaround.
Part of the problem might be the model geometry.  If walls are getting too thin to slice there will be issues with the print.  At .583 HHE the "bubble" looking smoke coming from the chimney are all separate pieces and that part of the model will absolutely fail.

Why are you going to 3 decimal places for HHE?  If you need a hole larger you should adjust the model.  There can be a lot of "unintended consequences" when you try to use a setting to adjust a model feature.

Moosiecakes, 2024-01-08:
> Thanks for the report. There appears to be something going on in regards to Hole Horizontal Expansion. When I set it to any non-zero number slicing appears to stop. If I turn on "Print Thin Walls" [...]

I have an STL that i use to Calibrate my printer so that pieces will fit properly into each other. The end result of Measuring the printed pieces seems to give me a HHE of 0.58375 unless I'm doing something wrong.  

here's the link. I've been using it on 3 of my printers. Parts seem to fit better, this is the first time I've encountered an issue. 

https://youtu.be/1O-Ho47rwLY?si=R5g9ZtFeoU5e_3Mf

GregValiant, 2024-01-08:
_"I have an STL that i use to Calibrate my printer so that pieces will fit properly into each other."_
I'm not saying there isn't a bug here (it looks like there might be) but trying to "tolerance" models using slicer settings isn't the right way to go about it.  The models should be designed to fit together, and then printed to exact specs.  If the OD of one model is 25mm and the ID of the next is 25mm then by definition they won't fit together because the features are "line to line" and would be referred to as a "press fit".  Pieces that are supposed to fit together must be designed to fit together.  Either the outside part should be designed with a larger hole, or the inside part with a smaller OD.  Giving any slicer "the model you have" and asking it to print "the model you want" rarely works well.

Moosiecakes, 2024-01-08:
I stumbled up on this video, it seemed feasible to me that this would work for me. To which it has,  being relatively new to 3d printing this seemed to be an option for me to get things to fit easier than they used to. I'm never looking for excellence, Just a good print if possible. The included model is one I've printed over xmas many times. I just thought it was odd that it wouldn't slice after I'd rechecked my values and inputted  a new number.  Being I'm pretty new at all this i wasn't aware if Cura could do this without the HHE being manipulated with a value.

GregValiant, 2024-01-08:
The model should have sliced without having to turn on Print Thin Walls.  It may not have crashed, but it was certainly taking it's time.
I was checking a model here last week and I walked away for a day and forgot I had Cura slicing the model.  When I finally came back, slicing had finished.  I checked the Cura log and slicing took 38 hours!!!.  It was a difficult little part and I scaled it up to pretty much fill my build space as a test, but 38 hrs?