Ultimaker/Cura #17931: Slicing bug? Spiralize outer contour active it will not slice any roof like the calibration cube

LightBridge16, 2024-01-07:
### Cura Version

5.6

### Operating System

macOS Monterey

### Printer

Creality CR-10S Pro V2 Ender Sprite Extruder Pro BLtouch 2.85 mm filament

### Reproduction steps

every model that has a roof will not sliced well when activating Spiralize outer contour

### Actual results

no roof only walls and support

### Expected results

closed model

### Add your .zip and screenshots here ⬇️

[CCR10SPRO_HOLDING_PLATE_MEADE_LIGHTBRIDGE.3mf.zip](https://github.com/Ultimaker/Cura/files/13854303/CCR10SPRO_HOLDING_PLATE_MEADE_LIGHTBRIDGE.3mf.zip)
[image]

Asterchades, 2024-01-08:
Correct. When using `Spiralize outer contour` - colloquially known as "Vase Mode" - the top is left open. This is by design, and is not only how the feature has functioned since it was first incorporated but is how the same feature functions in every other current slicer I know of.

GregValiant, 2024-01-08:
In addition to what Asterchades said, you can only spiralize one model at a time.  Any multiple models mean that there will be islands on layers and the toolpath will try to connect the islands.  The same thing happens with single models that devolve into islands.
You can see large red lines between your models.  Those aren't strings, they are extrusions.  You can also see many travels between the models.  Those don't happen when a model is correct to be spiralized.

I'll go ahead and close this as your setup is not compatible with "Spiralize Outer Contour and so what you see is "expected behavior".  I hope you understand.

closed 2024-01-08 by GregValiant

LightBridge16, 2024-01-09:
thank you for explaining it.

its a pity that this future is not available for somepart of the printing proces on a model to get that nice seamless look.

LightBridge16, 2024-01-09:
i have a model that is a GT2 gear beneed a cone with 5 cilinders onto of it. it would be super nice too make this futures possible for some parts of a printing model.

GregValiant, 2024-01-09:
Spiralize is a "walls only" process.  Without infill, a "roof" would always fail.
In order to have a seamless toolpath, the nozzle can never stop extruding from the top to the bottom.  The nozzle can never "cross air" because extruding would need to stop.  That always leaves a seam.  That is why "islands" are not permitted.

This model is a cone with a hole through it.
[image]

Turning on "Spiralize" will automatically turn on "Remove all Holes" and so the spiral model would not have a hole through it.  It will also not have a roof (which would be impossible to print).
[image]

With Spiralize on and Remove all Holes turned off you can see that when the nozzle moves from the outer wall to the wall around the hole, it keeps extruding until the center "island" is complete and then it stops extruding and "travels" back to the outer wall.  That is why there is a seam in this slice.
[image]