Ultimaker/Cura #17939: A model I have sliced before. Might be the combination or spacing.

GNiessen, 2024-01-08:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

FLSun V400

### Name abnormal settings

Nothing obvious.

### Describe model location

Mainly center. I tried moving and rotating.

### Describe your model

A small latch.

### Add your .zip here ⬇️

[SliceError.zip](https://github.com/Ultimaker/Cura/files/13856209/SliceError.zip)

GNiessen, 2024-01-08:
I tried it in Cura 5.4.0 and it worked.

cartman-sp69, 2024-01-08:
As mentioned since I have updated to version 5.6.0 I'm unable to slice any models I even went back to version 4.13 which was working and the same thing. I have completely removed all versions of cura and reinstalled same thing. it is not a model issue it is something with the application after the update.

GregValiant, 2024-01-08:
Thanks for the report.
I can't duplicate this.
Something must have changed within your computer if 4.13.1 doesn't work.  
I'm on Win10 Pro and this is your project with Cura 5.6.0.
[image]

GNiessen, 2024-01-08:
Sorry if I was not clear. It worked fine in 4.13.1 and 5.4.0, but not in 5.6.0. It doesn't indicate what the issue is, so I don't know what I can change.

closed 2024-04-03 by HellAholic