Ultimaker/Cura #17941: impression

dorianfpv, 2024-01-08:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

ender 3 s1 plus

### Name abnormal settings

suport

### Describe model location

oui

### Describe your model

case boitier ssd m2

### Add your .zip here ⬇️

[CE3S1P_Corps1.zip](https://github.com/Ultimaker/Cura/files/13859172/CE3S1P_Corps1.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

dorianfpv, 2024-04-05:
Merci de votre réponse sa fonction parfaitement bon développementde la 5.8.0 !