Ultimaker/Cura #17942: One At A Time fails when objects enlarged

benclewett, 2024-01-08:
### Cura Version

 5.6.0

### Operating System

MacOS 14.1.2

### Printer

Ender-3 V2

### Reproduction steps

Take a small object. (Sample attached.)  Create, say, 3 copies, and distribute with no overlap.  This will slice.  Then increase in size to 150%.  Despite the objects still having no overlap, the slice will fail with "Unable to Slice".

### Actual results

Error: Unable to Slice:
Please review settings and check if your models:
- Fit within the build volume
- Are assigned to an enabled extruder
- Are not all set as modifier meshes

### Expected results

A .gcode file.

### Add your .zip and screenshots here ⬇️

[woggle.zip](https://github.com/Ultimaker/Cura/files/13859886/woggle.zip)

GregValiant, 2024-01-08:
Thanks for the report.
The models in the project are 31.25 tall (125% scale) and your gantry height is 25mm.  The gantry height is the distance between the tip of the nozzle and the bottom of the X beam.  When the parts are 100% (25mm tall) the X beam can pass over a finished print and so One-at-a-Time will slice.  If a finished print is 31.25 tall then the X beam can smash into it so One-at-a-Time will not allow the slice.
This is expected behavior and is a safety to prevent crashes, so I'll remove the bug label and close this.

With only 3 parts you can cheat a little bit (if you dare).
Set them up like below and move them around so the left front prints, then the middle, and finally the right rear.  Go into the Machine Settings and increase the Gantry Height to 35mm.  It will slice.  because the X beam and print head keeps moving to the rear for each successive print, you should be OK (I'm not making any promises here - consider yourself warned).
Remember to set the gantry height back to 25 after this slice.  The number is in there as a safety.
[image]

closed 2024-01-08 by GregValiant

benclewett, 2024-01-08:
That makes perfect sense and I fully understand now.  Sorry for the ticket.

Thanks for the tip, I can see that if each part is Y+ from the last part,
by at least the Y dimension of the part, minus the Y distance between
the nozzle and gantry: then the gantry will never hit, great idea!!  (With
no skirt/raft/brim.)  I'll try that now...

(I presume they are ordered as shown in the part list.)

Ben

GregValiant, 2024-01-08:
_(I presume they are ordered as shown in the part list.)_
They are ordered more or less randomly.  You have to slice, check the preview, and go by that.  A little diagram on a piece of paper helps when you have a lot of parts.  For three it will be easy.

benclewett, 2024-01-09:
In my case they were printed exactly in reverse order.  I printed off 5 in
one go and it worked perfectly.

I think there is an optimisation in the program to organise the parts to
avoid gantry hits, for somebody who’s brave!

Thanks

Ben