Ultimaker/Cura #17952: Printing multiple small and one big item in sequence

Malaber, 2024-01-08:
### Is your feature request related to a problem?

I often print a lot of small object and one bigger one. When printing in sequence the build height is reduced to avoid collisions with previously printed objects. The last one thats prints would not have that problem though, as after that one, the nozzle just goes to the parking position.

### Describe the solution you'd like

When printing multiple objects in sequence I would like to select the order they are printed in or at least choose which one gets printed last (or the slicer automatically picks the biggest one?) so that a single large object can be printed with multiple smaller parts in sequence.

### Describe alternatives you've considered

/

### Affected users and/or printers

Everyone, especially people printing multiple objects.

5axes, 2024-01-11:
https://github.com/Ultimaker/Cura/pull/17893

T9Air, 2024-04-04:
A similar feature was added to version 5.7.0 beta, where if you select to print one at a time, you can choose to either have it choose the order or for the order to go by which model was imported first.

GregValiant, 2024-04-09:
This also made it into 5.7.0.
When you enable "One at a Time" a new command will appear right below it "Set print sequence manually".
With that enabled you can right-click on a name in the object list (lower left corner of the workspace) and move it up or down in the print order.

Take a look and let us know if this can be closed.

schmidt3d, 2024-12-20:
The core of this request is to allow printing many small and one large object in all-at-once-mode. This does not actually require for the user to be able to define the print sequence manually, so it's not(!) equivalent to [#17893](https://github.com/Ultimaker/Cura/pull/17893).

Let me try to explain again: The all-at-once-mode allows for printing multiple small objects, but they all are restricted in height to avoid collisions with the gantry.
Technically, this constraint could be relaxed by allowing exactly one large object which would have to be printed last. So Cura would arrange the print order such that all small objects are printed first (with height restrictions), and for the very last object these height restrictions are lifted. As the print sequence is done after the very last object there cannot happen any collision, so this procedure is safe.

GregValiant, 2024-12-20:
Don't you get the same effect by adjusting the gantry height?  If you set the tall model to print last there wouldn't be a problem with a collision.
My gantry height is 25mm.  Here I have adjusted it to 150mm and I've set the tall part to print last.

https://github.com/user-attachments/assets/8ea9922b-f42f-4e6a-b33c-1e294a940ae4

schmidt3d, 2025-01-02:
Hi [GregValiant](https://github.com/GregValiant), yes, your example looks like this should work, so I guess as a workaround it's ok. 
It puts more responsibility on the user, of course, because the collision avoidance is disabled this ways.

GregValiant, 2025-01-02:
There are always some complaints about the "gantry height" limiting the size of models in "One at a time" print sequence.  It is a safety setting and as such won't be going anywhere.  It can be worked around, but it does require some thought by the user.

There is (what I consider) a bug in "One at a Time".  Cura does not add a final "Z" move to the Transit Height to clear previous models that might be taller.  Because of that, at the end of the print the nozzle can hit a taller model if there is a parking move in the Ending Gcode (many machines have "G0 X0 Y{machine_depth} ; Present Print" in the Ending Gcode).
This is kind of hard to make out but that wide blue travel line goes right through the taller print in the left rear corner
[image]

I dabble in post-processors.  I call this one "Little Utilities" and it's a collection of 15 seldom used but often handy-when-you-need-them scripts.  Unzip it and put "Little Utilities v15.py" into the "scripts" folder within your Configuration folder.  It will be available with the other post-processors in Cura under "Extensions".
[LittleUtilities_v15.zip](https://github.com/user-attachments/files/18290599/LittleUtilities_v15.zip)

To use it: add it to the post-processing list and then enable "Cura Bug Fix Scripts" and then "One-at-a-Time Final Z".
[image]

The script reads the highest point of the print (the MAXZ) in the gcode.  That's the travel height between models in a One-at-a-Time print.  It adds a line to the gcode right at the end of the print but before the "Ending Gcode" runs.

In this example, the "MAXZ" of the print was 55.0mm because that back taller model is 50mm high.  This line insures that the nozzle will be above the tallest print when the nozzle moves to the "Present Print" position.

`G0 F2100 Z55.0                ; one_at_a_time final Z move`

That's why I call them _Little Utilities_.  Not often needed...but handy.

There are a couple of other scripts for One-at-a-Time.  If you "Renumber Layers" you can open the gcode in Cura and it will preview as a true "One at a Time" print.

I'll go ahead and close this.  Because the Gantry Height is an anti-collision safety feature, this won't be changed in Cura.  I hope you understand.

FYI the bug report for the final Z move is #10471.  It remains a problem in 5.9.0.

closed 2025-01-02 by GregValiant