Ultimaker/Cura #17958: Slicing Failed

lol4645, 2024-01-09:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3/ Ender 3 v2

### Name abnormal settings

Tree supports (everywhere)/ 20% infill/ Super quality Does not slice with default profiles

### Describe model location

Slices when I scale it down/ original scale - 300%, does not slice/ Smaller scale - 100%, Slices

### Describe your model

I have sliced it before

### Add your .zip here ⬇️

[Anatomically Correct Heart.zip](https://github.com/Ultimaker/Cura/files/13879039/Anatomically.Correct.Heart.zip)

5axes, 2024-01-11:
Model Error ?
[image]

GregValiant, 2026-09-27:
I'll go ahead and close this as being stale.

closed 2026-09-27 by GregValiant