Ultimaker/Cura #17959: Tree support Failing

sognatore99, 2024-01-09:
### Cura Version

5.6.0

### Operating System

Windows 11 Pro

### Printer

Ender 3 pro

### Name abnormal settings

No

### Describe model location

Model Located In center

### Describe your model

I have sliced it many times in version 5.3.0-alpha+xmas

### Add your .zip here ⬇️

If I slice in normal support mode, then switch to tree it works for standard profile. But then if you try to slice it again using tree it fails.  It fails if I slice in super quality mode no matter what with Tree Support.  I have slice this model many time in the 5.3.0-alpha+xmas without fail. I have attached the model and a recording of the issue.  Thank you.
[Issue.zip](https://github.com/Ultimaker/Cura/files/13879090/Issue.zip)

jellespijker, 2024-01-10:
It seems that you have **exported** the model instead of saving the project file. This means that we don't have all the used settings and materials. Can you save the project and upload that in a zip

sognatore99, 2024-01-10:
I have attached the project file(s).  Also of note is that this is the first time that I have used this version of Cura and my profiles did not come over like all the other times.  I had a difficult time understanding why cura was reporting "cannot access my profiles after changing nozzle size wrong quality type".  I managed to get it to work by following some other people's notes but not really sure what allowed the profiles to be eventually seen.  I chose a standard profile then a super profile from the list and not ones that I had modified.  I included three project files.  The very first time I opened Cura this morning the model with tree support sliced fine.  I then moved it a little and it failed.  The standard quality is one that I believe I did not modify any settings, but it could be a port from the older version, not sure.
[Tree_Fail_Projects.zip](https://github.com/Ultimaker/Cura/files/13886521/Tree_Fail_Projects.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

sognatore99, 2024-04-03:
So there is still a problem with my model.  I have two models, they look alike but have some slight differences.  Since it has been a few months since I worked on this I can not tell you what those differences are.  However, one model always fails when dropped in Cura.  Another test I did was to drop the failed model in then also dropped the good model, which I moved out of the way of the first model.  It sliced ok, which was weird, to me at least. I have included the STL files also.
[Cura_Slicing_Issue.zip](https://github.com/Ultimaker/Cura/files/14854845/Cura_Slicing_Issue.zip)