Ultimaker/Cura #17960: Creality CR 10 SE network connection problem

ayeginligil, 2024-01-09:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality CR 10 SE

### Reproduction steps

I want to connect my device, but my device is not in your system. Also, I cannot connect to the network. When I want to add an IP, it does not see it.

### Actual results

I want to connect my device, but my device is not in your system. Also, I cannot connect to the network. When I want to add an IP, it does not see it.

### Expected results

I want to connect my device, but my device is not in your system. Also, I cannot connect to the network. When I want to add an IP, it does not see it.

### Add your .zip and screenshots here ⬇️

[image]
[image]

Asterchades, 2024-01-10:
Cura's network connection facilities are for Ultimaker printers only. In order to connect to other brands you'll need to see if there's an appropriate plugin available on the Marketplace, such as the OctoPrint and Moonraker plugins.

My understanding is that the CR-10 SE runs Klipper firmware (whether it admits it or not), so likely requires the Moonraker plugin unless you're willing to do further, unnecessary configuration.

In terms of which definition to use, the SE is functionally identical to its predecessor as far as Cura is concerned so should work just fine using the regular CR-10 definition. Retraction is defined with the material properties so it won't make an appreciable difference that it no longer uses a remote extruder so long as the material you've chosen is properly defined. Note that this definition does not include any ABL commands in the Start GCode, so this will need to be added should you want it to run each time (or you can use a definition such as the CR-10S Pro which includes a G29 command).

GregValiant, 2026-09-27:
I'll close this.

closed 2026-09-27 by GregValiant