Ultimaker/Cura #17964: Adaptive layers cause "Stop at Layer / Height" to fail.

3Dkat, 2024-01-10:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Flying Bear Reborn 2 / Geeetech A10M

### Reproduction steps

When printing a multicolor items, I was chasing shorter print times, turning on adaptive layers did cut print times significantly, but at the cost of non working "pause at layer or z height". This issue had to be backtracked for several prints, since it was not so apparent, due to many different tweaks set in the settings at that particular time. Pretty much when I print multicolor objects, I set pause at layer number on the printer side and very rarely through Cura's "Post Processing" option, and everything works like a charm. Once I came across the adaptive layers function and turned it on, to my surprise, neither of the prior mentioned methods would pause my prints, even the manual editing of the g-code would work (Command m600 or m0). Took me a while to isolate and narrow it downs to the culprit that is 'Adaptive Layers" option. I tried using adaptive layers with many different other options on or off, and it has no effect unless you turn off that particular option.

### Actual results

Turning off Use Adaptive Layers, allows to execute pause at layer, change filament etc.. with no problems.

### Expected results

Turning off Use Adaptive Layers, allows to execute pause at layer, change filament etc.. with no problems.

### Add your .zip and screenshots here ⬇️

[FBG5_Hino_AdaptL_On.zip](https://github.com/Ultimaker/Cura/files/13886033/FBG5_Hino_AdaptL_On.zip)

GregValiant, 2024-01-11:
Thanks for the report.
Adaptive layers is meant for rounded, organically shaped models.  The layer height fluctuates to make smaller steps when printing something like a ball.  Your rectangular model has no domes and so Adaptive Layers goes right to the Max Layer Height as dictated by Maximum Variation, Variation Step Size, and Topography Size and it stays at 0.34 layer height once it gets above the Initial Layer.  That's why your print times are short.  You could just turn off Adaptive layers and increase your layer height to 0.34.

Within Pause at Height the code looks for "Initial Layer Height" and "Layer Height" and does some math to figure out what height a layer should be.  Adaptive Layers introduces different layer heights and so that portion of the code fails.  Z-hops in a file will also cause "By Height" to err as it will put the pause code in according to what it saw in a z-hop line rather than the "real" working height.

I have been re-working Pause at Height and if it is accepted, it will be "Pause at Layer" and there won't be a "By Height" anymore because it can be fooled by other settings.  Cura always changes heights at layer changes anyway.  For now, just use "By Layer" and use the Cura preview and enter the number of the last layer you want to print with the current color.

I'll leave this open for the Cura team.

GregValiant, 2026-09-27:
I'll go ahead and close this as being stale, and there are workarounds.  PauseAtHeight has been reworked and will be included with 5.14.0.

closed 2026-09-27 by GregValiant