Ultimaker/Cura #17966: Unable to slice, but no error code

upatamby, 2024-01-10:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ultimaker 3E

### Name abnormal settings

No

### Describe model location

Center. Moving does not fix the problem

### Describe your model

It's watertight

### Add your .zip here ⬇️

After opening cura and loading a model Cura starts slicing forever. If I move the model then the slicing stops because with the message: "Unable to slice with the current settings. The following settings have error:", but it shows no error code.
But, if I change a random setting then the slicing works fine, even if I change the setting and immediately change it back to the previous value. After applying this work around Cura keeps working fine, even if I remove the mpodel and load it again.
So it only (and always) happens immediately after a new start of Cura (also if I open a second instance of Cura).

This was a problem already in Cura 5.5.0
[slice error.zip](https://github.com/Ultimaker/Cura/files/13886929/slice.error.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

upatamby, 2024-04-03:
Thanks for looking at this issue. It still is not solved in my environment.
If I open Cura and load a model the automatic slicing starts and runs forever. If I move the model I get this error message without errorcode or indication which setting has an error. To be able to slice I have to make a random change in a random setting. After that it keeps working fine until the next time I restart Cura. Is there perhaps any ghost-setting active in the background? (Just guessing). Were you able to reproduce the issue? It is not dependent on the model. I tried several older and new models.

HellAholic, 2024-04-03:
Hey @upatamby ,
The model you shared doesn't even take 5 seconds to slice, which is confusing.
[image]

- Do you have any plugins installed? Remove all 3rd party plugins you've installed and try again.
- Can you disable the auto slice and try to slice manually, does it work then?
- If you don't have any specific profiles set, I recommend clearing your configuration folder and adding the printer again to see if that resolves the issue. [On Windows: %APPDATA%\cura\ -> Remove the 4.X or 5.X folders in there]

This does not seem to be a Cura issue but rather something on your device.
The slicing error is due to the automatic slicing trying to send data while either a setting is being changed or model is removed. So it's not a slice crash but an interruption in the process, but we don't have a separate message for that.

upatamby, 2024-04-04:
I uninstalled all add-ins. Same issue.
I disabled auto slice and tried to slice manually. This works fine. So I will leave auto slice off as a temporary work around.
I deleted all 4.x and 5.x folders and added my printer again. Still the same issue as soon as I enable auto slice.
Is there anything else I could try?

HellAholic, 2024-04-04:
We first need to find out why the auto slice is causing the issue for you, for now the best path forward would be using cura without auto-slice as you suggested.
To help us narrow down the cause of the issue, enable the auto slice so that it fails to slice, then go to the same configuration folder (cura\5.7) and find the cura.log file.
If you could find and upload your cura.log files that would be helpful as I cannot reproduce the issue on the machines I have access to (currently 4 different machines).
I'll reopen this issue and if you can provide the log file I'll add another ticket to investigate the issue with the log.

Thanks :)

upatamby, 2024-04-04:
[cura.zip](https://github.com/Ultimaker/Cura/files/14872319/cura.zip)

I did the following:
- delete configuration.
- start Cura and add printer again
- enable auto slice
- restart Cura
- drag a model into Cura
- slicing started at: 2024-04-04 15:47:52,265 (see log file)
- wait a few minutes (nothing happens)
- change a random property (support enabled)
- slicing finished in a few seconds
- restart cura
- drag model into Cura
- slicing started
- In preferences I disabled auto slice and immediately enabled it again. Slicing ended with error.
- I repeated this a few times
- Finally I changed a property (disabled support) and slicing finished normally within a few seconds

Thank you for all your effort. I hope you can find a solution.

HellAholic, 2024-04-04:
Special thanks for the detailed explanation and all the work you've put in making the report :)
I've added a ticket to discuss and investigate the logs.
For internal Cura team reference: CURA-11815

Carlos121084, 2024-08-27:
Hello All,
I have the exact same issue since around 1 year.
It happens only with auto slice.
And like @upatamby , if I change a random parameter, and remove that change 1 second after, then auto slice works fine until I restart the software.
Without auto slice, the slice button works with no settings modifications
Hope this can help on solving this (and for sure, that's clearly not a big deal)
Thanks

Ghostkeeper, 2024-09-11:
Yes, I've seen this issue too since version 5.6.0, and still in 5.8.1. It occurs only with auto-slicing enabled.

When the first model is loaded, any model, the slicing will stay on the indefinite progress bar (before the first progress message is sent back by CuraEngine to the front-end).
Changing the build plate then will cause the slicing to restart, but this time and any following times it will give an error message:
[image]

Changing any setting (and optionally, changing it back) will cause the slicing to properly start and slice successfully for the rest of time until Cura is restarted.

A relevant fragment of the log:
```
2024-09-11 18:33:44,070 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'cube.stl'>,trans.=<0.000,0.688,0.000>), took 1ms
2024-09-11 18:33:44,080 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.000,-9.632,0.000>), took 1ms
2024-09-11 18:33:44,134 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-09-11 18:33:44.134] [info] Connected to 127.0.0.1:49674
2024-09-11 18:33:44,148 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.028250694274902344 seconds
2024-09-11 18:33:44,605 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [331]: Creating thumbnail image (just before slice)...
2024-09-11 18:33:44,606 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [252]: Changing perspective mode to orthographic.
2024-09-11 18:33:44,609 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_Cura-5dHztz2/share/cura/resources/shaders/overhang.shader]...
2024-09-11 18:33:44,612 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_Cura-5dHztz2/share/cura/resources/shaders/transparent_object.shader]...
2024-09-11 18:33:44,613 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/tmp/.mount_Cura-5dHztz2/share/cura/resources/shaders/striped.shader]...
2024-09-11 18:33:44,716 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [348]: Starting to slice...
2024-09-11 18:33:44,716 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [368]: Going to slice build plate [0]!
2024-09-11 18:33:45,966 - WARNING - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [499]: Unable to slice with the current settings. The following settings have errors: 
```

MeNAce22AU, 2024-09-18:
I have the same issue.

Ghostkeeper, 2024-12-01:
This issue seems to be fixed in version 5.9. For me at least. @upatamby would you like to try again with the new release?

upatamby, 2024-12-02:
Hooray!!! The issue no longer occurs for me in version 5.9.0.
Thanks to everyone who contributed to the solution.

closed 2024-12-02 by upatamby