Ultimaker/Cura #17969: Slicing error when adding support blocker

banjug, 2024-01-10:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3 v2

### Describe your model

Error started when using support blocker

### Add your .zip here ⬇️

[CE3E3V2_Support_ONLY.zip](https://github.com/Ultimaker/Cura/files/13889811/CE3E3V2_Support_ONLY.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

banjug, 2024-04-03:
I've tested it out on my end as well, and I can confirm that I was able to slice the model successfully.
Thank you for the update!