Ultimaker/Cura #17970: Young White Dragon

IanJCL, 2024-01-10:
### Cura Version

5.6.0

### Operating System

Windows 10 Pro

### Printer

Dremel 3D45

### Name abnormal settings

Does not slice on default settings above Medium Quality. the model was designed for resign printers.

### Describe model location

Model does not slice regardless of orientation or location

### Describe your model

I have sliced this model at approximately 125% size with default Medium Quality settings. The model is watertight and Mesh Fix found nothing. 

### Add your .zip here ⬇️

[D_White_Dragon_Young_Updated.zip](https://github.com/Ultimaker/Cura/files/13889754/D_White_Dragon_Young_Updated.zip)

I am attempting to use settings for miniatures for this model, printing at low layer height and low speed.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

IanJCL, 2024-04-03:
It works! Thank you so much.