Ultimaker/Cura #17982: Slicing failed with an unexpected error

dagiiiii, 2024-01-11:
### Cura Version

5.6.0

### Operating System

Linux Mint 21.2 Cinnamon; Intel© Xeon© CPU E3-1231 v3; NVIDIA Corporation GF108GL [Quadro 600]

### Printer

Creality Ender-3 S1 Pro

### Name abnormal settings

the crash occures also on default profile

### Describe model location

both location as well as rotation is without meaning - crash always occures

### Describe your model

I can successfully slice the project on Cura 4.13.0

### Add your .zip here ⬇️

[CE3PRO_RocketFriede01F-Base.3mf.zip](https://github.com/Ultimaker/Cura/files/13901753/CE3PRO_RocketFriede01F-Base.3mf.zip)

5axes, 2024-01-11:
No issue for me on Windows 10 .
[image]

jellespijker, 2024-01-13:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release. I'm running on Manjaro Linux.

[image]

I'm closing this issue as fixed for the 5.7.0 release

closed 2024-01-13 by jellespijker