Ultimaker/Cura #17986: Crazy gCode generation: Upcounting extruder rate to far over E600 line by line

EmEiBee, 2024-01-11:
### Cura Version

5.5.0, 5.6.0

### Operating System

WIN10pro x64

### Printer

DIY (i3 based)

### Reproduction steps

Place a level test (https://www.printables.com/de/model/36706-bed-level-test) in CURA and resize to place on the bed (bed: 235x235, resize model to 200x200).
Slice with default "Super Quality"
Save gCode to file and open with a editor.

### Actual results

Don't try to print that! It goes fast into a massive exploding over extrusion !

### Expected results

A correct gCode

### Add your .zip and screenshots here ⬇️

[GI_3x3_cali_rl.zip](https://github.com/Ultimaker/Cura/files/13909081/GI_3x3_cali_rl.zip)
[image]
[image]

Asterchades, 2024-01-12:
There's nothing out of place in what you've provided here. Those E values aren't unexpected for the second layer (LAYER:0 indicates the first layer) of such a print, especially after over 8,800 lines and nearly 20 minutes, nor are they unexpected when Absolute Extrusion is enabled (or, more specifically, when Relative Extrusion is _not_ enabled, which is the default in Cura).

The only issues I see are with your Start GCode, which you've very clearly altered from the definition's default values. Without actually enabling `Relative Extrusion` in the Special Modes section of the print profile, Cura is still generating Absolute E values - which your printer is then interpreting as Relative by virtue of your having manually invoked an M83. Cura does auto-generate an appropriate M82 to enable Absolute Extrusion mode, however this comes _before_ your Start GCode where you override it.

You've also used incorrect placeholder names, likely from a Slic3r derivative (I suspect Orca), which is causing them to be passed as plain text and ultimately ignored (though the automatically generated temperature commands will be masking this for the most part).

To wit, you want to do the following:
- Enable `Relative Extrusion` under Special Modes or invoke an M82 at the end of your Start GCode
- Switch `[first_layer_temperature]` for `{material_print_temperature_layer_0}`
- Switch `[first_layer_bed_temperature]` for `{material_bed_temperature_layer_0}`
- Switch `[machine_max_acceleration_extruding]` for `{acceleration_print}`
- Switch `[machine_max_acceleration_retracting]` for `{acceleration_travel}`

closed 2024-01-12 by Asterchades

EmEiBee, 2024-01-12:
Good morning...
... and thank you very much for the detailed analysis and explanation.

To finally set the record straight...
I started 3D printing at the end of last year using an old Geeetech i3 in conjunction with Octoprint and CURA. At that point it worked out of the box without any problems.
I gradually changed and adapted the start code from the beginning and also used this code in OrcaSlicer after I switched to Repetier and I initially couldn't get the external plugin for CURA to work.
Orca and Cura therefore have an identical start script (I forgot to rewrite things like [first_layer_temperature] vs {material_print_temperature_layer_0} here), generally identical settings (copied by hand from CURAs SuperQuality profile) and also the same STL file.

If I understand this correctly, then CURA natively assumes absolute positioning on the extruder, while ORCA and SLIC3R assume relative positioning. Have I understood that correctly?
If so, then please forgive me for seeing an error here. Because I wasn't aware of this difference (until today)...

Greetz
Micha

Asterchades, 2024-01-12:
I'm honestly not sure what Orca assumes as I've never used it. My knowledge on its functionality is only rudimentary at best.

My experience with its "parent" slicers - Slic3r and PrusaSlicer - suggest that Absolute would still be the default. Prusa even goes so far as to use a faux relative mode when requested as it requires you to insert a script to reset the E state at the beginning of every layer (though why they didn't program it to do this automatically instead of bringing up a warning asking you to do it yourself is beyond me) and then each layer is still processed using Absolute values. The intent is to minimise cumulative extrusion errors from rounding, but when you're going down to micrometre measurements the steppers themselves will already be more inaccurate than any rounding issues.

Orca could well have a different default, either globally or for whatever you've based your definition on, or its parser used on the starting GCode may have been set up such that it detects the presence of an M83 and adjusts accordingly.

But, yes, Cura most definitely defaults to absolute extrusion values for _nearly_ every definition included in the base package. The Cremaker and Image3D definitions are the only ones which buck this trend, by virtue of enabling the previously mentioned Relative Extrusion setting. Most definitions work around this by performing any purge functions with absolute values before resetting E at the end (G92 E0), ready for the print.