Ultimaker/Cura #17987: [FR] Change Start and Stop GCODE depending on profile/material

sargonphin, 2024-01-12:
### Is your feature request related to a problem?

--

### Describe the solution you'd like

Hello,

I am aware that this is a duplicate of an already existing issue but I would like to open this one to enhance the interest into this feature

I am currently rebuilding a printer with Marlin and I have setup printing profiles in Cura, except that I need different start and stop gcode depending on the profile

The important feature that I need is the ability to change "M913 - Set Hybrid Threshold Speed" in Marlin which sets the speed at which the printer goes from stealthChop (silent but slow) to spreadCycle (noisy but fast). Because of TMC technical limitations, relying on a fixed threshold can lead to failed prints. It needs to be tailored depending on the target print and travel speeds

Alternatively, there are other control GCODE such as different values for Input Shaping (M593), Linear Advance (M900) and TMC chopper settings (M919) that could be different depending on the target speeds

### Describe alternatives you've considered

It is possible to do it manually, of course. Or to create different printers with different start/stop gcodes depending on the target print speeds

### Affected users and/or printers

Printer builders that benefit from Cura as a slicer

### Additional information & file uploads

--

GregValiant, 2024-01-12:
Cura 5.6 includes the ability to do math and some logic in the StartUp and Ending Gcodes.
This is from my own startup.  If I'm printing PLA I don't wait for the bed but if I'm printing PETG then I want to wait for the bed.
{'M190 S' if material_bed_temperature_layer_0 > 65 else 'M140 S'}{material_bed_temperature_layer_0}   ; Wait if over 65

This one holds the purge line speeds down.
G1 F{speed_print*60*2/3} X1.0 Y200.0 E15 ; First purge line

As long as the target speeds don't change then you wouldn't have to go in and change the lines.
{'M913' if speed_print > 100 else ';M913'}

closed 2025-04-05 by sargonphin

GregValiant, 2025-04-05:
Cura 5.9 (and 5.10) have changed to a slightly different format.  It's more like writing code.
Again, this is from my startup. and changes my steps/mm depending on the material.

{if 'TPU' in material_type}
M92 E119                   ; Steps/mm for TPU
{elif 'PETG' in material_type}
M92 E102                   ; Steps/mm for PETG
{else}
M92 E97                    ; Steps/mm for other materials
{endif}

sargonphin, 2025-04-05:
You responded to all my tickets I can see 😁😁 Yes I am aware this is why I closed this ticket as completed :)