Ultimaker/Cura #17988: Slice Failure

sidewinderzz, 2024-01-12:
### Cura Version

5.6.0

### Operating System

win11

### Printer

Ender3 S1

### Name abnormal settings

Tree supports, 70mm/sec print speed, cubic subdivision infill

### Describe model location

orientation changes naught what ails it

### Describe your model

yes i have sliced and printed b4

### Add your .zip here ⬇️

[Bug Report.zip](https://github.com/Ultimaker/Cura/files/13910565/Bug.Report.zip)
Everything slices fine until i select cubic subdivision as the infill pattern. I've had this issue more than once so idt its a fluke.

jellespijker, 2024-01-13:
It seems like you exported the 3mf, which makes it hard for us to reproduce the issue. Can you save the project instead. This will store a 3mf file with all the relevant settings for the materials and printers

sidewinderzz, 2024-04-03:
> We made some improvements to prevent slicing fails in Cura 5.7.0 available
> to download now. I was able to slice your model successfully in 5.7.0.
> Could you test if that's the case for you as well?
> You can download it here:
> https://github.com/Ultimaker/Cura/releases/tag/5.7.0

Awesome I'll be checking it out... I've had quite a few slicing failures
recently, so thanks for the update

HellAholic, 2024-04-03:
This is the first couple of steps towards rooting out the slicing crashes.
We've included sentry reporting to get data on the reason behind slicing crashes.
So even if your tricky models fail to slice, it will give us more accurate data to prevent those in future releases.

GregValiant, 2026-09-27:
I'm going through these old reports.  Cura 5.6 and 5.5 suffered from problems with certain models. 5.7 was much better.
I'll go ahead and close this.

closed 2026-09-27 by GregValiant