Ultimaker/Cura #17991: slicing engine failed

schoubert, 2024-01-12:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 Max

### Name abnormal settings

no and no

### Describe model location

no

### Describe your model

yes with a previous cura version

### Add your .zip here ⬇️

i have message " slicing engine stop to process" during slicing
[Uploading Buste ok 2024.zip…]()

schoubert, 2024-01-12:
[image]

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc