Ultimaker/Cura #17997: Add MingDa Magician X2 printer

rbeede, 2024-01-12:
### Is your feature request related to a problem?

Missing the MingDa Magician X2 as a printer

### Describe the solution you'd like

Add the MingDa Magician X2 as a printer

### Describe alternatives you've considered

Using the vendor's forked version of Cura

### Affected users and/or printers

Users with the Magician X2 printer
[Mindga_X2_cfgs.zip](https://github.com/Ultimaker/Cura/files/13924245/Mindga_X2_cfgs.zip)

rbeede, 2024-01-12:
[image]

rbeede, 2024-01-12:
[image]

rbeede, 2024-01-12:
Start G-code

```
 G28 ; home all axes
 M420 S1
 M117 ; Purge extruder
 G92 E0 ; reset extruder
 G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
 G1 X2 Y20 Z0.3 F5000.0 ; move to start-line position
 G1 X2 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
 G1 X2 Y200.0 Z0.4 F5000.0 ; move to side a little
 G1 X2 Y20 Z0.4 F1500.0 E30 ; draw 2nd line
 G92 E0 ; reset extruder
 G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
```

rbeede, 2024-01-12:
End G-code
```
 G91; relative positioning
 G1 Z1.0 F3000 ; move z up little to prevent scratching of print
 G90; absolute positioning
 G1 X0 Y0 F1000 ; prepare for part removal
 M104 S0; turn off extruder
 M140 S0 ; turn off bed
 M84 ; disable motors
 M420 S0
 M106 S0 ; turn off fan
```

rbeede, 2024-01-12:
Might be the same as the Mingda Magician X.  Perhaps update the existing built-in X to say  X/X2  ?

LilBub, 2024-01-16:
I have this machine as well.  I just followed their steps to manually add the machine.  But, it is also convenient if it can be added to future builds of Cura

mckunkel, 2024-02-05:
> I have this machine as well. I just followed their steps to manually add the machine. But, it is also convenient if it can be added to future builds of Cura

Greetings, can you elaborate on what you mean by "their steps to manually add this machine"

Asongalo, 2024-08-02:
yeah I looked and there's a Mingda Magician X but no Mingda Magician X2

Asongalo, 2024-08-02:
I looked and there's a Mingda Magician X but no Mingda Magician X2 for some reason