Ultimaker/Cura #18000: Windows on Arm build for Cura

mikednonotthatmiked, 2024-01-12:
### Is your feature request related to a problem?

I have an Arm-based Windows laptop (yes, they do exist!) and would like to use it with Cura for PC-directed printing. I can run the x86-64 build, but at a tremendous penalty to both performance and battery life. A native Arm build would resolve those issues.

### Describe the solution you'd like

I'd like an Arm native build on Cura for Windows. This would put Arm based hardware for Windows on par with Apple Arm hardware.

### Describe alternatives you've considered

Running an x86-64 build on Arm is possible, but at a large performance penalty.

### Affected users and/or printers

This could also benefit Arm users for Linux, who also do not have a native binary version available from Ultimaker.

lexcyn, 2024-06-05:
Bumping up this request since there are a ton of new ARM Windows systems coming out this month and it would be nice to have native ARM support.

diamondhoodgem, 2025-02-02:
Bumping up as well as finding running an x86-64 build on Arm is verry slow and cant slice most models

matthewharrington, 2025-06-02:
Yes please. Also here for an ARM build.

swagstaff1972, 2026-01-07:
Bumping up, been almost a year now since original request

spirilis, 2026-01-13:
Bumping up - Turns out the mouse behavior, particularly right-click and drag to change perspective, is very unstable using the x86_64 Cura build with Windows 11 ARM.  It often doesn't work and shows the right-click menu instead of twisting perspective, and randomly shows the right-click menu in the middle of movement even when it does work.  I'm curious if a native build will behave correctly here.

Version: Cura 5.11.0
Windows 11 ARM64 25H2 build 26200.7462