Ultimaker/Cura #18003: Incorrect object slicing, deep square open container

goofyseeker311, 2024-01-13:
### Cura Version

5.6.0

### Operating System

Win11 RTX3080

### Printer

elegoo neptune 3 pro

### Reproduction steps

slicing a square open top container

### Actual results

slicing output is whole inside filled with grid support even if its empty and no top

### Expected results

should be bottom and four walls, no supports, only adhesion

### Add your .zip and screenshots here ⬇️

[image]
[EN3PRO_akkuprojekti12a.zip](https://github.com/Ultimaker/Cura/files/13927783/EN3PRO_akkuprojekti12a.zip)

GregValiant, 2024-01-14:
Thanks for the report.
When I open the 3mf file I get this model.  The red surfaces facing upward are inverted surfaces.  They are a problem because a slicer can't tell the inside from the outside.  The model needs to be repaired.
[image]

That 3mf file is not a project file so I have to guess, and I'm guessing you might have "Remove all Holes" turned on.  The model errors can also cause a model to have false walls and/or tops/bottoms but the main culprit is usually "Remove all Holes".

I'll take the bug label off this for now.  Let us know if turning off "Remove all Holes" or repairing the model makes this problem go away.
It didn't show up when I sliced the model.
[image]

goofyseeker311, 2024-01-15:
so i just need to make sure the surface normals are correct. hmm.

closed 2024-01-15 by goofyseeker311