Ultimaker/Cura #18011: Slicing Failed

adoeller, 2024-01-13:
### Cura Version

5.6.0

### Operating System

Windows

### Printer

Tevo Tornado

### Describe your model

sucessfully sliced before.

### Add your .zip here ⬇️

[TT_Clip Lichterkette v2.zip](https://github.com/Ultimaker/Cura/files/13929027/TT_Clip.Lichterkette.v2.zip)

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc

adoeller, 2024-04-03:
I switched to orca slicer (and an other printer)