Ultimaker/Cura #18016: Slincing failed on 5.6, while everything works on 5.4

mroelandts, 2024-01-14:
### Cura Version

5.6.0

### Operating System

Ubuntu 22.04.3 LTS

### Printer

Creality Enter-3 Pro

### Name abnormal settings

No, same settings as with the 5.4 release

### Describe model location

center

### Describe your model

Nothing special

### Add your .zip here ⬇️

[CE3PRO_jan_2_battery.zip](https://github.com/Ultimaker/Cura/files/13931675/CE3PRO_jan_2_battery.zip)

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc

mroelandts, 2024-04-07:
Issues are still present in 5.7.0