Ultimaker/Cura #18018: Extruder End GCode isn't taken into account

woodencase01, 2024-01-14:
### Cura Version

5.6.0

### Operating System

Win 11

### Printer

Climber 7, Marlin

### Reproduction steps

1. Configure a custon machine with 2 extruders
2. Add gcode to "End GCode" for each extruder
3. Slice the part

### Actual results

The custom GCode for the extruder End GCode isn't in the file

### Expected results

Extruder GCode should be present at the end of the file

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-01-14:
I have a fix for that in this group of post processors I call "LittleUtilities".  Unzip the file and put it into the "scripts" folder inside your Configuration Folder (use "Help | Show Configuration Folder" to find it).
It will be available with the rest of the post-processors under "Extensions / PostProcess / Modify Gcode" and then "Add a script".
The one you want is the second one down in the list "Add Last Extruder Ending Gcode".
You obviously aren't the first to report this bug.  I'll mark this as a duplicate and hopefully it will be fixed in 5.7.0.
[LittleUtilitiesI_GV.zip](https://github.com/Ultimaker/Cura/files/13932918/LittleUtilitiesI_GV.zip)

See @rburema , these things do come in handy.

rburema, 2024-01-18:
I never said they weren't :-)

GregValiant, 2026-09-27:
I'm going through these old reports and closing the stale ones.  I believe this problem was fixed.

closed 2026-09-27 by GregValiant