Ultimaker/Cura #18019: Impossible Jigsaw Puzzle failed to slice. worked in the past.

AlecHolcomb, 2024-01-14:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

Ender 3

### Name abnormal settings

doesnt work on default profiles either

### Describe model location

moved from center did not work. rotations did not work either

### Describe your model

jigsaw puzzle, sliced correctly before

### Add your .zip here ⬇️

[puzzle.zip](https://github.com/Ultimaker/Cura/files/13932458/puzzle.zip)

TheAtomSmasher, 2024-01-16:
I've been having this issue as well - files that sliced and printed in the past are now failing. I'm probably going to just load an older version of Cura until they fix it.

AlecHolcomb, 2024-01-16:
> TheAtomSmasher

Which Version worked well for you in the past? I just got into printing.

TheAtomSmasher, 2024-01-16:
I honestly don't know, life got in the way and I haven't printed anything in over 2 years, so it would be a pretty old version. In my experience, you don't have to go back very far to avoid massive glitches because they get fixed quickly.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!