Ultimaker/Cura #18020: Slicing crashes when Support Top Distance is less than layer height

Morganiron, 2024-01-14:
### Cura Version

5.6.0

### Operating System

Windows 11 Home, Version 23H2,  Processor: 12th Gen Intel(R) Core(TM) i5-12400F 2.50 GHz, RAM: 16.0 GB

### Printer

Ender-3

### Name abnormal settings

Slicing started getting stuck around 50% after adjusting support settings, specifically changing from "Everywhere" to "Touching Buildplate." Slicing crashes after changing the support Z distance from 0.2 to 0.12

### Describe model location

Center of buildplate, rotated 180 degrees from original placement to face forward on the buildplate.

### Describe your model

I have sliced this model several times before. I scaled it up a lot since the original was a miniature.

### Add your .zip here ⬇️

[CE3E3V2_Kittencorn.zip](https://github.com/Ultimaker/Cura/files/13932803/CE3E3V2_Kittencorn.zip)

I was playing with the settings to try to get the best quality/time/material balance for this print and get the support structures as close as possible while still making them easy to remove with minimal scarring.

I know this isn't a critical error as I'm sure most people wouldn't make this same adjustment.

It seems the only setting that is causing the crash is changing the Support Z distance (and therefore Support Top Distance) from 0.2 to 0.12 (or anything less than 0.2(layer height). 

- The model will slice with a smaller Support Z Distance as long as I ensure Support Top Distance >= layer height.

GregValiant, 2024-01-14:
Thanks for the report.
I can duplicate this with your model but not with any other.
Cura will not split layer heights just for a support interface gap.  What I've seen is if a gap is smaller than 1 layer height, CUra makes the gap "0".  If the gap is just under 2 layer heights Cura will make the gap 1 layer height.  I think there is something else going on here besides the obvious change in the Z gap over the interface.

The Cura team will take a look.

Vandresc, 2024-03-01:
For our internal reference CURA-11249/11424.

GregValiant, 2026-09-27:
I'm closing these old duplicate reports.  5.5 and 5.6 had a lot of problems.  99% were fixed in 5.7.  The current version is 5.13.0.

closed 2026-09-27 by GregValiant