Ultimaker/Cura #18023: model wont slice

Rader9609, 2024-01-14:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

Ender 5 S1

### Reproduction steps

I added a model and clicked slice

### Actual results

got an error

### Expected results

the model slices

### Add your .zip and screenshots here ⬇️

[FILE.zip](https://github.com/Ultimaker/Cura/files/13933196/FILE.zip)
[image]

mlandry71, 2024-01-15:
I fixed mine by removing the CuraEngine Tiled Infill Generation plugin.

Cysign, 2024-01-15:
When Cura can't slice, sometimes it helps changing the brim width i.e. from 1.0 to 1.1 and back to 1.0.

saumyaj3, 2024-01-15:
@Rader9609 Hey! I was trying to look into the model but the zip file is empty can you please send the model again? Thanks :)

closed 2024-01-29 by github-actions[bot]