Ultimaker/Cura #18048: Cura 5.6.0 is not adjusting speed properly, resulting in zits/bumps on prints

moccor, 2024-01-16:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3 V3 SE

### Reproduction steps

Go to the speed settings in 5.6.0 and make changes to all the settings, 5.6.0 "Print Speed" will not be grayed out. However on 5.4.0 when you make changes to varying speeds, it does gray out, informing us that it is not using a fixed "Print Speed" but is using the settings from everywhere else and ignoring "Print Speed". Resulting in prints with zits/bumps that look similar to "random Z seam". I have ruled out everything and tried everything, the problem all corrected once I tried a version older than 5.6.0 and noticed this

### Actual results

Zits on prints looking like random z seam, when using 5.6.0

### Expected results

No zits on prints when using the same exact hardware, software and profiles on 5.4.0

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-01-17:
Your screenshot doesn't match your description, making it hard to say for certain what is happening.

Parent settings, such as `Print Speed`, are only disabled (greyed out) when all of their children are given an explicit value (indicated by the "ƒx" marker). This is the case in your 5.4 settings, but your 5.6 settings still leave `Wall Speed` (even if `Outer Wall Speed` has been set) and `Top Surface Skin Speed` as being calculated. Whether that calculation involves the `Print Speed` or not doesn't matter - just that there are still child settings being left without a user-specified value in the UI.

With that said, the only values which vary between the two of them are the new Top Surface Wall speeds (which will have previously been the regular Wall speeds). Everything else matches. Since everything prints using its specified value (the global `Print Speed` setting is only used for calculations - no feature specifically uses this speed) this means that the speeds actually being used by Cura for the resulting print match between the two versions outside of those two specific settings.

As to what's actually causing your problem? I'm unable to answer that with any certainty given the information provided. The V3 SE did not have a definition in 5.4 (this was only added with 5.6) so I'm unsure which base definition you've used, and the provided images don't point to anything which could be responsible. A Project File from both versions would go a long way towards determining what may (or may not) have changed between the two.

moccor, 2024-01-17:
I added a "Ender 3 V2" and then just changed the dimensions and settings to match the V3 SE into 5.4.0 and everything works fine.

As for the settings "not matching" I did all that I could, those settings aren't available in 5.4.0 like they are in the 5.6.0 screenshot. I doubt it is those settings though, since the bumps/zits happen entirely throughout objects. 

Myself and others on reddit have experienced problems similar to this and I encouraged them to try 5.4.0. I spent multiple days and many hours troubleshooting this, I didn't really feel like going and uploading zips and files and figuring out where to go to do it. Many software will just have a button that is like "download log zip", or maybe they should just automatically do that when clicking "report a bug" to make this process easier. I had hoped the screenshot showing the circled differences would be enough, cuz I used the same exact profiles in both. Same hardware, same software (5.6.0 vs 5.4.0), same STL, everything the same besides Cura versions. And 5.6.0 has zits/bumps

LukisG, 2024-01-19:
For me I don't have that problem speed atjust just fine via USB and gcode in SD, tested individually each setting making it 10mm/sec for example on first layer speed it works just fine, same with others.

Printer profile E3 v3 se,
Filement settings PLA and Petg (tested on both) 
Profile setting changed tempreture and speed.

moccor, 2024-01-19:
Just gonna copy and paste this here too cuz you wanted to post a bad test in two spots :

Sounds like you used speeds that were so slow that it wouldn't mess up like I had explained without it adjusting properly. What I meant was, "Ender 3 V3 SE Creality speeds", which was what was in my screenshot. Anything printed slow will look amazing and perfect. Another important thing is the object, simple objects will also look fine because they won't even use much varying speeds

Haxiboy, 2024-03-20:
I experience the same, Cura 5.6 + Klipper makes zits/blobs on curved surfaces. Same settings/same everything on Cura 5.4 is not. I've had the same problem in 5.3.X as well...seems like this problem is re-occurring time to time.

GregValiant, 2026-09-27:
It's been 2 1/2 years.  I'll go ahead and close this.

closed 2026-09-27 by GregValiant