Ultimaker/Cura #18054: Regardless of the print temperature I set in Cura (usually around 200° celsius), the printer will heat to 175°

oricoh, 2024-01-17:
### Cura Version

5.6.0

### Operating System

MacOS Sonoma

### Printer

Ender 3 Pro

### Reproduction steps

1. I prepare and slice a print in Cura (any print)
2. I set the print temperature to 205° 

### Actual results

I start the print, but the printer shows that its warming up to 175° only (instead of 205°)

### Expected results

The printer should warm up to 205° as instructed in Cura

### Add your .zip and screenshots here ⬇️

[175.3mf.zip](https://github.com/Ultimaker/Cura/files/13961486/175.3mf.zip)
[image]

oricoh, 2024-01-17:
Just to add that, what I do afterwards is to tune and increase the temperature manually in the printer to the correct temperature.

closed 2024-01-17 by GregValiant

oricoh, 2024-01-17:
Thanks, I'll try it. Just wondering why this behaviour started now, as I have not experienced it before Cura 5.6.0.... any thoughts?

GregValiant, 2024-01-17:
The printer definition files come from either a member of the community, or from the printer manufacturer.  Someone likely submitted a pull request to change the startup gcode in the definition file and when you upgraded to 5.6 your printer grabbed the new definition.  There is no way that UltiMaker can go out and buy one of every printer out there so things like this (a minor change the the startup gcode) often gets approved without another owner of the same printer model looking at it.
I would have approved this one if I would have been called on because it is minor.  

If you liked the way it was you can go back to the older Cura version and create a project file.  Open that project file in Cura 5.6 and when the printer is imported as part of the project, the old startup gcode will come with it.

I always need to keep a special definition file for my own printer because I open so many project files here and if I'm distracted my custom startup can (and does) get over-written.  I open up one of my own project files and that straightens everything out.