Ultimaker/Cura #18059: Auto Scaling Button

ZackyDB, 2024-01-18:
### Is your feature request related to a problem?

When I import a custom modal for some reason it auto scales it to be bigger.

### Describe the solution you'd like

It would help so that when I import a custom modal it will leave it alone and let ME do the scaling.

### Describe alternatives you've considered

It could be an on and off switch in the settings.

### Affected users and/or printers

I will benefit from this and maybe other people will too.

Asterchades, 2024-01-18:
There are two existing settings in the General Cura preferences which read as "Scale large models" and "Scale extremely small models" where the latter (I believe) defaults to being enabled. Do these settings not already do what you're requesting?

ZackyDB, 2024-01-18:
Oh I am so sorry I didn’t see that 😅 Thanks for letting me know!😁