Ultimaker/Cura #18062: Slicing shutting down after severals minutes (10-15min)

ThibaultRad, 2024-01-18:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Describe model location

Centred

### Describe your model

A large bin with details

### Add your .zip here ⬇️

[Perforated chassis Radiall.zip](https://github.com/Ultimaker/Cura/files/13975561/Perforated.chassis.Radiall.zip)
We carried out several tests reducing the accuracy of the model on Cura 5.6, then tested it on Cura 5.4 and 4.6. The results are shown in the .txt file in the compressed folder.
You will find a Parasolid file (open with SolidWorks then export to .stl for Cura), the .3mf file for Cura 5.6 and the file for Cura 4.6.

ThibaultRad, 2024-01-18:
The compressed folder : [Perforated chassis Rad.zip](https://github.com/Ultimaker/Cura/files/13975614/Perforated.chassis.Rad.zip)

GregValiant, 2026-09-27:
I'm going through and checking on these old reports.  I'll close this one.

closed 2026-09-27 by GregValiant