Ultimaker/Cura #18067: SHOW CONFIGURATION FOLDER DOES NOT OPEN

SadikHakanGuris, 2024-01-18:
### Cura Version

5.6.0

### Operating System

WINDOWS 10 PRO 64 BIT 

### Printer

ANYCUBIC KOBRA  2 MAX 3D PRINTER  

### Reproduction steps

 ON ULTIMAKER CUDA 5.6 WHEN WE GO TO THE SHOW CONFIGURATION FOLDER 
[image]

THE FOLDER DOES NOT OPEN  

THE SYSTEM 

PROCESSOR AMD RYZEN 9 5900X 
INSTALLED RAM 64 GB 
SYSTEM TYPE 64 BIT OPERATING SYSTEM 
WINDOWS 10 PRO 22H2 

### Actual results

THE RESULT IS THE SHOW FOLDER DOES NOT OPEN AND CAN NOT UPLOAD THE CONFIGURATION TO THE SOFTWARE 
[image]

### Expected results

IT SHOULD BE OPEND AND UPLOAD THE CONFIGURATIONS TO THE SOFTWARE OPENED

### Add your .zip and screenshots here ⬇️

[image]
[image]
THESE ARE THE SCREENSHOTS

GregValiant, 2024-01-19:
Thanks for the report.
This is a duplicate of another report ( #16466 and [CURA-11026](https://ultimaker.atlassian.net/browse/CURA-11026)) and I commented there as I had this problem as well.  I never did fix it on that computer (running Win 10 Home) but my new laptop with Win10 Pro doesn't have the problem.

The Cura team will take another look.

GregValiant, 2026-09-27:
I think this problem was found and fixed.  I'll close this.

closed 2026-09-27 by GregValiant