Ultimaker/Cura #18072: Slicing error after making model taller

Lula-kitsune, 2024-01-18:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Snapmaker 2 A350 Dial extruder

### Name abnormal settings

I have some overrides but doesn't print when set to default

### Describe model location

no on either

### Describe your model

Before making it taller it sliced fine as soon as i made it taller to meet requirements it started failing. object was checked with meshtools and is water tight.

### Add your .zip here ⬇️

[cura slicing error.zip](https://github.com/Ultimaker/Cura/files/13982606/cura.slicing.error.zip)

GregValiant, 2024-01-22:
I coudn't get it to slice either.
The Cura team will take a look.

HellAholic, 2024-02-16:
Sorry it took us a while to get back to you,
the issue should be resolved in the next release, please keep an eye out and re-open if you still experience the issue.
[image]

closed 2024-02-16 by HellAholic