Ultimaker/Cura #18073: 5.4.0 onwards  "Unhandled exception in script" failed to execute script 'curra_app'

deno5555, 2024-01-19:
### Cura Version

5.4.0 onwards

### Operating System

windows 8.1

### Printer

anycubic 

### Reproduction steps

Cura 5.3.1 and earlier works but from ultimaker cura 5.4.0 and onwards it comes up with
 "Unhandled exception in script" failed to execute script 'curra_app'. 
Traceback (most recent call last):
  File "cura_app.py", line 28, in <module>
  File "PyInstaller\loader\pyimod02_importers.py", line 352, in exec_module
  File "cura\CrashHandler.py", line 24, in <module>
    from PyQt6.QtWidgets import QDialog, QDialogButtonBox, QVBoxLayout, QLabel, QTextEdit, QGroupBox, QCheckBox, QPushButton
ImportError: DLL load failed while importing QtWidgets: The specified procedure could not be found.

### Actual results

Traceback (most recent call last):
  File "cura_app.py", line 28, in <module>
  File "PyInstaller\loader\pyimod02_importers.py", line 352, in exec_module
  File "cura\CrashHandler.py", line 24, in <module>
    from PyQt6.QtWidgets import QDialog, QDialogButtonBox, QVBoxLayout, QLabel, QTextEdit, QGroupBox, QCheckBox, QPushButton
ImportError: DLL load failed while importing QtWidgets: The specified procedure could not be found.

### Expected results

cura program starts up.

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2024-01-19:
Windows 8 is not longer supported:
https://support.makerbot.com/s/article/1667337917781

To support newer operating systems, Cura 5.4 switched to using version 6.4 of the Qt framework, and that version of Qt no longer supports Windows 8.

Unless you can update to Windows 10 or 11, Cura 5.4 may be the latest version you can run.

fieldOfView, 2024-01-19:
I am closing this issue because Windows 8 is not a supported operating system (not even by Microsoft), so unfortunately this is a fact of life.

closed 2024-01-19 by fieldOfView