Ultimaker/Cura #18075: Tree Support causes crash

BlueSteel04, 2024-01-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

Tree support

### Describe model location

Middle of plate

### Describe your model

Yes, not with Tree

### Add your .zip here ⬇️

[CE3E3V2_Solar Panel ARM v1.zip](https://github.com/Ultimaker/Cura/files/13986093/CE3E3V2_Solar.Panel.ARM.v1.zip)
It's not only with this model. Sometimes I can change the tree slightly to make it work, but I have not been successful this time.

5axes, 2024-01-21:
Classic tree support error on versions 5.5 & 5.6 test on 5.7 Alpha  : https://github.com/Ultimaker/Cura/discussions/18080

[image]

BlueSteel04, 2024-01-22:
> Classic tree support error on versions 5.5 & 5.6 test on 5.7 Alpha : #18080 [...]

Thank you!

closed 2024-01-22 by BlueSteel04