Ultimaker/Cura #18076: [0.5.6 Bug] Between tree supports and interface there is a funny layer

Sophist-UK, 2024-01-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Dagoma DiscoEasy 200

### Reproduction steps

1. Load a model requiring tree supports
2. Slice
3. Check the slicing for 1 layer below interface

### Actual results

[image]

As can be seen, there is a very sparse (possibly non-tree) support layer printed between the tree supports and the interface layer.

### Expected results

Tree supports should reach right up to the interface layer.

### Add your .zip and screenshots here ⬇️

[image]

[Connector.zip](https://github.com/Ultimaker/Cura/files/13988162/Connector.zip)

Sophist-UK, 2024-01-19:
Changing the `Support Pattern` (which is irrelevant and should probably not be shown when tree supports are enabled) does NOT change this strange layer - so it doesn't seem to be a phantom non-tree support layer but rather a bug in the interface layers.

GregValiant, 2026-09-27:
That was intended to help the tree support interface stick better.  It is still in Cura.
I'll closet this.

closed 2026-09-27 by GregValiant