Ultimaker/Cura #18079: Infill "leaks" out of a manifold solid

etherbrat, 2024-01-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3 S1

### Reproduction steps

1. Imported an STL file
2. Sliced it

### Actual results

The object is a cylinder with holes in the wall.  It looks fine in the Prepare view.
The STL file is manifold according to both Cura 5.6.0 and Meshmixer 3.5.474.
The STL file is also manifold according to my own programmatic tests (by reading and analysing the file).
1. When sliced, the Preview view shows that central hole in the cylinder is full of infill, i.e. infill has somehow "leaked" out of the cylinder wall.
2. Cura 5.6.0 takes a long to to slice the file compared with Cure 4.6.0. Both Versions produce the same unexpected result.

### Expected results

Cura 5.6.0 should not "leak" infill from the cylinder wall into into the central hole of the cylinder.
PrusaSlicer slices the same STL file as expected, i.e. produces a cylinder with holes in the wall, and no infill in the central hole.
I cannot use Cura to slice this file.

Two project files are attached: One is a small-ish subset of the other. Cura produces the same unexpected result with both,

### Add your .zip and screenshots here ⬇️

[Drum.zip](https://github.com/Ultimaker/Cura/files/13988801/Drum.zip)

smartavionics, 2024-01-19:
I used the Mesh Tools plugin to fix the normals and now it slices OK for me.

[image]

smartavionics, 2024-01-19:
I should add that I couldn't use your profile so the print time isn't comparable to what you have.

etherbrat, 2024-01-20:
Oh, really … Well, that’s embarrassing (blush!). I read somewhere (obviously the wrong-headed opinion of a mixed up commentator) that “Cura doesn’t use normals”.Is there a view that’ll give me a hint about which normals are wrong?Much appreciated. Many thanks!NeilNeil Higgins ***@***.***

Screenshot_2024-01-19_13-39-51.png (view on web)

smartavionics, 2024-01-20:
The prepare view looks different. I think the red surfaces are considered to be underside?

Before normals fixed...

[image]

After normals fixed...

[image]

etherbrat, 2024-01-23:
Blender will draw facet normals. To my shock and horror, it showed that a lot more of them were wrong than I expected. Maybe because I’m left handed?

 

Screenshot_2024-01-20_08-56-57.png (view on web) <https://github.com/Ultimaker/Cura/assets/585618/66956a51-2fef-4a17-a6d6-232f228014bd> 

Screenshot_2024-01-20_08-57-39.png (view on web) <https://github.com/Ultimaker/Cura/assets/585618/af7f0e30-3c17-412e-be95-3c80fa6857e1>

GregValiant, 2026-09-27:
I'm closing these old duplicate reports and this is one of them.

closed 2026-09-27 by GregValiant