Ultimaker/Cura #18082: Slicing failed

KimABQ, 2024-01-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Voron 2.4r2

### Reproduction steps

Trying to slice a basket design and receiving an unknown error.

### Actual results

Failed slice,

### Expected results

Slice to complete as requested.

### Add your .zip and screenshots here ⬇️

[Failed slice.zip](https://github.com/Ultimaker/Cura/files/13991196/Failed.slice.zip)

5axes, 2024-01-19:
An other **Print Thin Walls** Issue. 
[image]

Same issue on 5.7 Alpha

GregValiant, 2024-01-19:
As downloaded and sliced with 5.7 alpha 1 on a Win 10 Pro laptop:
Backend exited abnormally with return code 3221225477!
With Print Thin Walls turned off:
Backend exited abnormally with return code 3221226505!

_BUT_ if I open the project and immediately change Support Placement from "Touching Buildplate" to "Everywhere" the slice finishes.  Switching back to "Touching Buildplate" the slice fails again.
[image]
@Vandresc 
@MariMakes

GregValiant, 2026-09-27:
I'm closing these old duplicate reports.  5.5 and 5.6 had a lot of problems.  99% were fixed in 5.7.  The current version is 5.13.0.

This is a report on that model:
-> Analyzing file
**--> 3 Naked edges  (?)
--> 1 Planar holes  (?)**
--> 0 Non-Planar holes  (?)
**--> 14 Non-Manifold edges  (?)
--> 18 Inverted normals  (?)
--> 28 Duplicate faces  (?)
--> 61 Degenerate faces  (?)**
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 3324
--> Vertex count changed from 7243 to 8740 **(+1497)**
--> Triangle count changed from 14521 to 17516 **(+2995)**

closed 2026-09-27 by GregValiant