Ultimaker/Cura #18083: New Bottom/Top Infill Patterns

radry, 2024-01-19:
### Is your feature request related to a problem?

It's 2024 and cura still only has 3 primitive infill patterns for top and bottom, while having a lot of (mostly redundant) internal infill patterns.

### Describe the solution you'd like

More Top and Bottom infill patterns, for example hilbert curve.
Other examples are sinus, small zig-zag, circles,... there are many possibilities.

### Describe alternatives you've considered

Using a different slicer.

### Affected users and/or printers

Everyone

kimber-tech, 2024-04-22:
Please add this feature!
I use this a lot for aesthetic prints, but have to change to another (un-preferred) slicer, would be awesome if I could do this in Cura.
Surely not hard to add patterns that are already in the software?

Loughty, 2024-10-08:
I would love to see this come true in Cura, not only does it change the look, but also the texture, which helps in a practical way in some prints (not just as an aesthetic).

Here are some examples of patterns:
[image]
_(Image credits Luca2210)_