Ultimaker/Cura #18085: Support Z distance doesn't leave an entire layer between since 5.6.0.  Perhaps intended behaviour ????

doubletrouble023, 2024-01-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

VZ-330

### Reproduction steps

Printing at 0.2layer height and have the Z-distance set to 0.1, always done it like that without any issue. So I won't have to think about making sure yet another parameter is correct for every slice...

Project file
[Cura_sup_test.zip](https://github.com/Ultimaker/Cura/files/13995543/Cura_sup_test.zip)

### Actual results

Video:
https://github.com/Ultimaker/Cura/assets/70795685/3a1044c1-55a2-4cf4-9b57-05b1f68d283b

[image]

### Expected results

0.2mm gap between model and support

### Add your .zip and screenshots here ⬇️

Project File:
[Cura_sup_test.zip](https://github.com/Ultimaker/Cura/files/13995562/Cura_sup_test.zip)

Video:
https://github.com/Ultimaker/Cura/assets/70795685/3a1044c1-55a2-4cf4-9b57-05b1f68d283b

[image]

Asterchades, 2024-01-20:
This was a deliberate change made with 5.6, though the UI doesn't really reflect what's happening if you don't know how to look at it.

> Updated the Support Z Distance to allow for more different values, no longer only multiple layerheights, solves https://github.com/Ultimaker/Cura/issues/6105

Because of how the preview works, however, Cura doesn't show thinner layers as being thinner - it shows the extrusion being narrower instead. You can see this by looking at the visible width of the line or by using the Flow preview type. So in this case the very top layer of the supports will be showing with only half the width, or half the flow, of all those below it, which ultimately results in it only being 0.1mm high instead of 0.2mm. _Note that this isn't simple under-extrusion as the Z value is also correct, even if the preview doesn't show it._

doubletrouble023, 2024-01-20:
Okay, I suspected it might be like that. Too bad this change wasn't communicated clearly(?) 
Because I ended up with an entire plate of failed prints because of this. Had no idea the supports would be impossible to remove

So if I understand you correctly, the movement in Z axis just for the last layer of support is 0.1 instead of 0.2? (in my case)
Then there really should be a 0.1mm gap there? Curious that it sticks like crazy then. I'll have to take a look what's going on during a test print, probably possible to see it in another gcode viewer as well.

Thanks

Asterchades, 2024-01-20:
That would be correct. It's fairly amazing how much difference as little as even 0.04mm can make to how well the model adheres (or not) to the supports, so I'm not at all that surprised that the difference between 0.2mm (a full layer) and 0.1mm is enough to stop them separating altogether.

In terms of a GCode viewer that will show it properly, I do know that Prusa's viewer shows it; though there's doubtless others.

doubletrouble023, 2024-01-21:
Alright, yeah I was thinking that  .1  gaps works fine when printing at that layer height, but ofc  a 0.2layer will droop more and therefore adhere better to the support.
Anyway, thanks for the help in clearing this up!

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant