Ultimaker/Cura #18101: Horizontal Expansion rendering incorrectly through infill

Robdoc, 2024-01-22:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Tevo Tornado

### Reproduction steps

Put file on workspace

add horizontal expansion

add supports

Note that the model looks like normal size when viewed through supports but not when viewed directly.

### Actual results

The horizontal expansion isn't rendered, the normal model size is shown through the transparency of infill.

### Expected results

Model should look same but when viewed directly and when through the supports.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

Asterchades, 2024-01-23:
Just wanting to double check as reading your description didn't make a lot of sense at first...

What you're reporting is a graphical issue. Horizontal Expansion is being correctly applied to the slice and is ultimately reflected in the Preview when viewed with just the model. However when the Supports, which are rendered partially transparent, pass in front of it in the viewport, any part of the model which is obscured by them is now rendered at the original model size.

That being the case I can indeed confirm this is happening. The type, size, and pattern of support and the model being obscured appears to be irrelevant - it simply needs to pass in front of the model when viewing the Preview.

Robdoc, 2024-01-26:
> Just wanting to double check as reading your description didn't make a lot of sense at first...
>
> What you're reporting is a graphical issue. Horizontal Expansion is being correctly applied to the [...]

Yep, just a graphical issue is what i'm reporting. The print was successful and looked like one would expect with horizontal expansion (when not viewed through the supports)